jpiat/arduino #1: use of enum ‘receiver_state’ without previous declaration error in the LiFiReceiver.ino

bsainandan, 2015-11-26:
I tried your code on my Arduino UNO board, when i dumped LiFiEmitter it is working fine on my Arduino board and when i dumped LIFIReceiver.ino to my Arduino UNO  i am getting this following error
use of enum ‘receiver_state’ without previous declaration. Can you  resolve me with this issue . Thanks in advance.

jpiat, 2015-11-26:
can you try again ? I just did a push.

lawrenciumbc, 2015-11-26:
I have just tried it now on a project I am working on and appear to have the same issue. Here is a copy of my error messages in case they are useful to you:

Arduino: 1.6.5 (Windows 7), Board: "Arduino Uno"

lifi_transiver:78: error: use of enum 'receiver_state' without previous declaration
lifi_transiver_reciever:53: error: conflicting declaration 'char frame_buffer [38]'
lifi_transiver:63: error: 'frame_buffer' has a previous declaration as 'unsigned char frame_buffer [38]'
lifi_transiver_reciever:54: error: conflicting declaration 'int frame_index'
lifi_transiver:64: error: 'frame_index' has a previous declaration as 'char frame_index'
lifi_transiver_reciever:55: error: conflicting declaration 'int frame_size'
lifi_transiver:65: error: 'frame_size' has a previous declaration as 'char frame_size'
lifi_transiver_reciever.ino: In function 'void setup()':
lifi_transiver_reciever:203: error: redefinition of 'void setup()'
lifi_transiver:145: error: 'void setup()' previously defined here
lifi_transiver_reciever.ino: In function 'void loop()':
lifi_transiver_reciever:218: error: redefinition of 'void loop()'
lifi_transiver:158: error: 'void loop()' previously defined here
lifi_transiver_reciever:236: error: no matching function for call to 'add_byte_to_frame(unsigned char [38], char*, char*, receiver_state*, unsigned char&)'
lifi_transiver_reciever.ino:236:107: note: candidates are:
lifi_transiver.ino:78:5: note: int add_byte_to_frame(char*, int*, int*, int*, unsigned char)
lifi_transiver.ino:78:5: note:   no known conversion for argument 2 from 'char*' to 'int*'
lifi_transiver_reciever.ino:167:5: note: int add_byte_to_frame(char*, int*, int*, receiver_state*, unsigned char)
lifi_transiver_reciever.ino:167:5: note:   no known conversion for argument 2 from 'char*' to 'int*'
lifi_transiver_reciever:238: error: call of overloaded 'println(unsigned char*)' is ambiguous
lifi_transiver_reciever.ino:238:40: note: candidates are:
In file included from C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Stream.h:26:0,
                 from C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/HardwareSerial.h:29,
                 from C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Arduino.h:223,
                 from C:\Users\NetLab\Documents\Arduino\libraries\TimerOne/TimerOne.h:21,
                 from lifi_transiver.ino:1:
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:71:12: note: size_t Print::println(const String&) <near match>
     size_t println(const String &s);
            ^
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:71:12: note:   no known conversion for argument 1 from 'unsigned char*' to 'const String&'
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:72:12: note: size_t Print::println(const char*) <near match>
     size_t println(const char[]);
            ^
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:72:12: note:   no known conversion for argument 1 from 'unsigned char*' to 'const char*'
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:73:12: note: size_t Print::println(char) <near match>
     size_t println(char);
            ^
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:73:12: note:   no known conversion for argument 1 from 'unsigned char*' to 'char'
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:74:12: note: size_t Print::println(unsigned char, int) <near match>
     size_t println(unsigned char, int = DEC);
            ^
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:74:12: note:   no known conversion for argument 1 from 'unsigned char*' to 'unsigned char'
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:75:12: note: size_t Print::println(int, int) <near match>
     size_t println(int, int = DEC);
            ^
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:75:12: note:   no known conversion for argument 1 from 'unsigned char*' to 'int'
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:76:12: note: size_t Print::println(unsigned int, int) <near match>
     size_t println(unsigned int, int = DEC);
            ^
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:76:12: note:   no known conversion for argument 1 from 'unsigned char*' to 'unsigned int'
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:77:12: note: size_t Print::println(long int, int) <near match>
     size_t println(long, int = DEC);
            ^
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:77:12: note:   no known conversion for argument 1 from 'unsigned char*' to 'long int'
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:78:12: note: size_t Print::println(long unsigned int, int) <near match>
[... cut ...]
C:\Users\NetLab\AppData\Roaming\Arduino15\packages\arduino\hardware\avr\1.6.2\cores\arduino/Print.h:78:12: note:   no known conversion for argument 1 from 'unsigned char*' to 'long unsigned int'
Multiple libraries were found for "TimerOne.h"

 Used: C:\Users\NetLab\Documents\Arduino\libraries\TimerOne

 Not used: C:\Users\NetLab\Documents\Arduino\libraries\TimerOne-r11

use of enum 'receiver_state' without previous declaration

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

bsainandan, 2015-11-27:
Thank you for your reply @jpiat now i successfully dumped my code. The error was enum has to be given in the separate header file. Arduino IDE will not accept this typedef in the program.  @lawrenciumbc try this, so i created separate include "types.h" in the program with the enum lines now its working. @jpiat now code is dumped but i am not able to see anything on the serial monitor? how exactly we have to do to get output on the serial monitor after dumping the code.

jpiat, 2015-11-27:
Do you have the emitter lighting the led (what color) ? Do you have the
receiving led wired in parallel with a 1Mohm resistor and wrired (with
correct polarity) between you ADC channel  (which one) and gnd ?
I have no problem with typedef in Arduino 1.6.6.

jpiat, 2015-11-27:
Demo code is setup for Analog input 3 to be used.

2015-11-27 7:08 GMT+01:00 Jonathan Piat <piat.jonathan@gmail.com>:

bsainandan, 2015-11-27:
On the transmitter side i have connected RED LED to the Digital 13 pin and dumped the emitter code LED blinks only once after code dumping, on the receiver side i have connected RED LED in parallel with 1 M ohm resistor to the A3 and other terminal of the LED to the GND and dumped receiver code.  Now i open the serial monitor on the receiver side and wait for the response, but i dont get any and i tried typing on transmitter serial monitor but there is no response on the receiver serial monitor. The baud rate i have set 1200. Can you please tell me how to proceed after connections and dumping the code.

jpiat, 2015-11-27:
For the emitter to use D13, you need to uncomment :

//These defines are for a LED connected on D13
/*#define OUT_LED() DDRB |= (1 << 5);
#define SET_LED() PORTB |= (1 << 5)
#define CLR_LED() PORTB &= ~(1 << 5)
*/

and comment the definition of the same macros some lines after.

Baudrate is setup for 115200bps.

bsainandan, 2015-11-27:
Now i did that modification and i see my transmitter LED is glowing now continuosly its an improvement thank you, but at the receiver code should i make any changes, because even now i don't see anything on the serial monitor and i also tried typing something on the serial monitor of transmitter and i don't see anything on the receiver screen. My receiver has this configuration LED +ve is connected to A3 and a resistor in parallel to the LED +ve  and the other terminal of the resistor, LED -ve are connected to the GND. IS this config correct or should we do something else. Resistor is 1 Mohm.

jpiat, 2015-11-27:
Wiring should now be good. You mentioned using a RGB led, what channel of
the LED are you using for emitter and what channel are you using for
receiver ? Is it a focused led ? Is it diffused or clear ?

bsainandan, 2015-11-27:
I am using RED LED for both emitter and receiver, It is a normal low power LED.We are placing the emitter(LED) above the receiver(LED) to get the light focused.When we changed the symbol duration on the transmitter end we could observe the change at the emitter(LED) .Whereas there is no test to confirm about the working of receiver. After making the connections at the receiver side  and dumping the code how can we say that the receiver is working? will i get some message on the serial monitor of the receiver? At now the emitter is fine but the receiver is not able to receive the message.

jpiat, 2015-11-27:
You shoudl alter the emitter rate as emitter and receiver are synchronized
on the same rate. The emitter keeps transmitting the same symbol sequence
so when the emitter and receiver are lined up correctly you should get the
message on the receiver side. Depending on the receiver led type, you may
need to try different distances (~6 to 15cm) and see if you get anything on
the receiver side. I'll make a quick modification to the code so you can at
least see the the raw bytes as they come.

jpiat, 2015-11-27:
I you have a oscilloscope available you can also try to measure the signal
o nthe receiver side to make sure you get something from the emitter.

2015-11-27 10:20 GMT+01:00 Jonathan Piat <piat.jonathan@gmail.com>:

jpiat, 2015-11-27:
I have added a DEBUG symbol to the programm that allow to turn on/off the
raw output of the receiver.

2015-11-27 10:21 GMT+01:00 Jonathan Piat <piat.jonathan@gmail.com>:

bsainandan, 2015-11-27:
I have tried with oscilloscope i see that some square waves on the screen from the emitter, so something is  coming from the emitter and receiver is not able to receive and process the signal. and i will check ur debug code and come back.

bsainandan, 2015-11-30:
@jpiat  I see that i am getting some voltage on  the multi meter and also on the oscilloscope but there is no response on the serial monitor of the receiver section, i did n't receive any message, the receiver circuit is also correct because i am getting some voltage, so i think the problem is with the code, i don't see anything on the screen and more over the debug code that you gave me is getting dumped into the arduino board but it is not loading inside the board(the LED with  Letter L on the board is not glowing when i dump debug code.). Kindly troubleshoot me this issue.

jpiat, 2015-11-30:
I don't understand what happen when loading the debug version in the
Arduino ? Do you use a genuine Arduino or a clone ? If using a close can
you point me to the clone version ? I tried this morning and the code just
load fine. I updated the code this morning for easier synchronization, can
you try again and report if this works for you ?

bsainandan, 2015-11-30:
@jpiat  when i am loading even your present version of the code to the board that LED with Letter L on the board is not glowing even now it means program is dumped but board is not able to load, my board is Arduino UNO original model. It is NOT Cloned Model. and in the latest code what is PinMode(13, output) should i connect something to it?

jpiat, 2015-11-30:
The L LED is not supposed to be lit up after connecting. This led does not
indicate a good loading but is controlled by program. If you have a serial
connection and DEBUG enabled at beginning of the code, you should get an
output when pointing the first led at the second one. Can you share a
picture of your setup ? I don't understand what the problem could be.

jpiat, 2015-11-30:
If youd 'ont have any serial output, please make sure that you are
configured for 115200bps on the console side.

2015-11-30 15:59 GMT+01:00 Jonathan Piat <piat.jonathan@gmail.com>:

bsainandan, 2015-11-30:
what do you mean by serial connection enabled? After some time i will send you the video of my setup.

bsainandan, 2015-12-01:
[image]
[image]
[image]
 I am sending you the pictures of my circuit 1st one with bread board is the receiver connections resisitor and LED in the parllel and green is given to the A3 and yellow is given to GND(shown in 2nd image), in the 3rd image is the LED glowing on the emitter after dumping the code.

jpiat, 2015-12-02:
i just made a push with a small change to display a start message in the
receiver terminal, if you don't see the message on startup this mean that
the baudrate is not configured properly.
Its very hard to understand the receiver led wiring in your pictures, but
even with a mis-wired led you should see something in the terminal

bsainandan, 2015-12-02:
@jpiat  I am able to see  your push message on the screen of the receiver serial monitor, and receiver circuit is LED and resistor(LED+ve and resistance one terminal are connected in the same vertical holes of the bread board and LED -ve and other terminal resistance are in the same vertical hole of the bread board) are connected parallel and green wire from the LED is given to the A3 and Yellow is given to the ground. Now i am able to see the message on the screen so what may be the problem that is stopping to receive the data.

bsainandan, 2015-12-02:
@jpiat After running a set of times i see that some times only some part of the println is getting printed (eg : Start of rece, instead  of Start of receiver program) i don't know what is happening to the remaining letters and also i gave some println statement inside the void loop() like Serial.println("Inside the loop") and forever it is running inside the loop, so what may be causing the problem? please help with this and also can you explain me everything how you have connected recent one 1w LED(both transmitter and receiver), i will try to replicate your circuit and see if its working or not.

jpiat, 2015-12-02:
The loop code, with DEBUG enabled, should print values as soon as data is
received from the emitter. When running the code and shinning the emitter
straight at the receiver, you don't get anything ?
Before wiring the 1W led, we should try to get it to run with the basic led
... What version of the Arduino IDE are you using ? With the latest version
there is a nice "Serial plotter" mode that allows to plot data sent by the
arduino. I can mod my code to enable a mode where the Arduino sends the
analog values it reads from the LED to understand what goes wrong.

bsainandan, 2015-12-03:
When I try to debug the code with Println statements i see that in the inside the void loop(), the if (new_word  == 1) is not getting executed so i wrote else statement for it and the program directly goes to the else statement because new_word is  not becoming 1. I have tried some sample analog read program with my receiver circuit connection and  when i shine the emitter i see the receiver is able to receive the values and i am able to see the change in the values depending on the shining. So with the same circuit connection when i dump your receiver code its not showing the output? The problem is i dump the code and shine emitter onto the receiver i don't see anything on the screen(only start of the... or with debug code i see only blank serial monitor), i am using latest arduino IDE 1.7.8 version i have. So you can mod your code and i will try to dump and see how it will run on my version.

jpiat, 2015-12-03:
I'am using 1.6.6 that is latest stable. I have added a DEBUG_ANALOG option
that will dump the analog values to the terminal so you can use the serial
plotter to see the result.  To enable this option you need to disable the
DEBUG symbol and enable the DEBUG_ANALOG symbol.
A problem that can occur with the blue led is that it generates greater
level  output and it saturate the ADC that uses the 1.1V reference. What
you could try is to shine the first blue led at a meter from the receiver.
If you happen to have red LED, you can also try those.

bsainandan, 2015-12-03:
I don't see any graph on my screen, but the serial monitor prints 1023 value continusly,,, do you know what that is?

jpiat, 2015-12-03:
1023 means that the ADC is saturated ... I will post a commit in a sec to
try to increase the range of the ADC.

jpiat, 2015-12-03:
I just pushed a commit. If you comment the line :

#define INT_REF

the ADC will switch to the 5v reference instead of the 1.1v internal.

2015-12-03 9:52 GMT+01:00 Jonathan Piat <piat.jonathan@gmail.com>:

bsainandan, 2015-12-03:
[image]

This is the output I got thank you man , you nailed it , now how to enhance this for a good results

jpiat, 2015-12-03:
Great ! This is the expected output. You cannot expect the transmission to
be super stable because of ambient light, led alignement and led problems
when used as a receiver.
To enhance transmission one user replaced the receiver led with a proper
light sensor.

lawrenciumbc, 2015-12-03:
@bsainandan I tried your suggestion below:

"The error was enum has to be given in the separate header file. Arduino IDE will not accept this typedef in the program. @lawrenciumbc try this, so i created separate include "types.h" in the program with the enum lines now its working."

but I keep getting a fatal error 'no such file or directory'. I have added '#include <types.h> ' to the top of my reciever file and pasted the enum lines into it. The file is sat in the same directory as the emitter and receiver folder, but I also tried 'sketch>add file' and played around with trying to add it as a zip library. 

I'm sure its something very basic and obvious that I have not done to link the include (I am new to Arduino which is probably why i'm getting this error). Do you have any ideas?

jpiat, 2015-12-03:
> '#include ' to the top of my reciever file and pasted the enum lines into

I really don't understand why this problem arise on your side and not on
mide. I'am using the official IDE version 1.6.6 downloaded from
https://www.arduino.cc/en/Main/Software and the software i have on this
github compiles with no error.

jpiat, 2015-12-03:
I think i understand the problem and it could be all because of the
arduino.org vs arduino.cc thing. Both have different IDE version and the
compiler might not exactly be the same.

2015-12-03 14:51 GMT+01:00 Jonathan Piat <piat.jonathan@gmail.com>:

bsainandan, 2015-12-03:
@jpiat Yeah i will try with Photosensor, and how did you drive 1w LED with the arduino board? will arduino drive 1w LED, i think it will burn? so how did you drive 1w LED and connect it to arduino .. will the same code work for that.

bsainandan, 2015-12-03:
@lawrenciumbc  Try what jpiat has suggested https://www.arduino.cc/en/Main/Software download windows version and try to compile code and see? for me i just pasted the code on the arduino tab and then press ctrl + shift + N and save it as Type.h and pasted the enum part in it and on the program i gave #include "Type.h" thats all and it worked for me.

bsainandan, 2015-12-07:
@jpiat  Hello i have updated LED receiver with the proper photo detector and everything is working fine but how to increase message length i see only 32 bytes of the data i am able transfer

jpiat, 2015-12-07:
Cool ! Did you have to make any change to the code ? What is the max
transmitting distance ?
The payload is limited to 32bytes (text data only for now, but can easily
evolve to raw bytes), because the receiving setup cannot guarantee that you
can keep the synchronization for long. If you want to transmit more data
you'll need to split into multiple 32bytes chunk.

bsainandan, 2015-12-08:
@jpiat  no i didn't make any change to the code it was working cool, with the photodetector and distance i am able to reach more than 60cm , i f i use good detector then it may increase, my only concern now is how tp increase the text length, if you can show me how to change the program to add multiples of 32 bytes chunk i will try that basically i am not from coding back ground.

closed 2015-12-12 by jpiat

amirezz, 2016-01-01:
hi all
thanks a lot for that project 
i just have the same error and want to have the types.h library to complete compiling the code 
thanks in advance

amirezz, 2016-01-01:
i just download arduino IDE v1.6.7 and the code working fine thanks all i will now try again

amirezz, 2016-01-07:
thanks a lot . it is working now and i have hello world on the serial monitor . now i am trying to have best results by using 12 v led bulb . and mosfet . now i have good emitter ( transmitter with continuously streaming white light ) i am wondering about the suitable baud rate & distance of that transmission . also i would to ask about the receiver side. what is the kind of sensor needed to receive white light ? is it need special circuit to amplify the received signal . thanks in advance @

jpiat, 2016-01-08:
You could use a photo-diode or photo transistor but its not tested. The
best way to achieve white light and transmit data is to use RGB led for
transmitter.

amirezz, 2016-01-08:
thanks for your interest and reply . but 1 watt RGP led have 3 m range only and also is sensitive to the orientation issue ( don't know that from the code or the led itself) i need to use normal led bulb and already make transmitter circuit which continuously transmit data  but can't receive data any idea about modify the code ( baud rate or arduino pin no ) . thanks in advance

amirezz, 2016-02-05:
any updates .
thanks in advance

jpiat, 2016-02-05:
the comminication speed can be configured using the symbol period define in
both the emitter and receiver. The emitinpin can be defined in the emitter
but you need to design the OUT_LED SET_LED CLR_LED macro. if you telle me
pins i want to work withh i ll add them to the files.