jpiat/arduino #2: Hello

mahmutdur, 2016-01-23:
I  run your receiver project but always made this error
LiFiReceiver.ino:47:86: error: use of enum 'receiver_state' without previous declaration

what is that 

Thank you

jpiat, 2016-01-23:
This problem occurs if you use the arduino llc ide instead of
<http://arduino.org>arduino. <http://arduino.org>cc ide (
https://www.arduino.cc/)

I run your receiver project but always made this error
LiFiReceiverino:47:86: error: use of enum 'receiver_state' without previous
declaration

what is that

Thank you

greengsm, 2016-02-01:
How to compile your code to use examples. For me the same error

jpiat, 2016-02-02:
Please use the arduino.cc IDE to compile the code :
https://www.arduino.cc/en/Main/Software

amirezz, 2016-02-09:
Thanks Jpiat for the update code it is really very helpful one .I wondered can i make the msg came from serial port( write on serial monitor at the emitter side and see the received message on the receiver arduino board serial monitor on anther pc)  
thanks in advance

jpiat, 2016-02-09:
This is already implemented. On the emitter side, if you uncomment the
define :

//#define TRANSMIT_SERIAL

the input data will be received on the serial port.

amirezz, 2016-02-13:
thank you very much .it works

amirezz, 2016-02-18:
dear jpiat thanks for your helpful code and also for you quick support . now i have that project working very well .but i need some Clarification to can explain that project as it is module هin  my graduation project and need to explain the code also .
my question in receiver side about (&(frame_buffer[1]))   why i must use ( & ) and 1 are it pointer  of
 fram_index or frame _ size or anther pointer else 
thanks in advance

jpiat, 2016-02-18:
&(frame_buffer[1]) is because the first character of the frame buffer is
the STX indicator which is not printable. Doing
 frame_buffer[1], takes the second character of the frame_buffer (start of
the text message) and adding the & permits to recover the pointer on this
second character since println expect a char * for its argument. What is
your graduation project about ? Did you make any addition to the code ?

amirezz, 2016-02-18:
my graduation project about emergency interrupt of the Traffic Signal Lights (i.e ambulance sends light message to the traffic signal lights to set the red  sign and after pass the traffic man make it green again )
so i modified that code as when receiver  side have that message emrg (emergency) it set red sign and the traffic man send message go to set it green again 
so i make that on code
 
if(((&(frame_buffer[1]))&&0x65)&((&(frame_buffer[2]))&&0x6D)&((&(frame_buffer[3]))&&0x72)&((&(frame_buffer[4]))&&0x67));//mean emrg in hex
     { digitalWrite(red sign , HIGH);
digitalWrite(green sign , LOW);
digitalWrite(yellow sign , LOW);}
     
  
     if(((&(frame_buffer[1]))&&0x67)&((&(frame_buffer[2]))&&0x6F));//mean go
     { digitalWrite(red sign ,LOW);
      digitalWrite(green sign , HIGH);
      digitalWrite(yellow sign , LOW);}

but it is not working very well some times hanging .and need to make every traffic signal light group addressable because it is make the two traffic group set the same sign .any idea  thanks in advance

amirezz, 2016-02-18:
my graduation project about emergency interrupt of the Traffic Signal Lights (i.e ambulance sends light message to the traffic signal lights to set the red sign and after pass the ambulance make it green again )
so i modified that code as when receiver side have that message emrg (emergency) it set red sign and the ambulance send message go to set it green again
so i make that on code

if(((&(frame_buffer[1]))&&0x65)&((&(frame_buffer[2]))&&0x6D)&((&(frame_buffer[3]))&&0x72)&((&(frame_buffer[4]))&&0x67));//mean emrg in hex
{ digitalWrite(red sign , HIGH);
digitalWrite(green sign , LOW);
digitalWrite(yellow sign , LOW);}

 if(((&(frame_buffer[1]))&&0x67)&((&(frame_buffer[2]))&&0x6F));//mean go
 { digitalWrite(red sign ,LOW);
  digitalWrite(green sign , HIGH);
  digitalWrite(yellow sign , LOW);}

but it is not working very well some times hanging .and need to make every traffic signal light group addressable because it is make the two traffic group set the same sign .any idea thanks in advance

amirezz, 2016-02-21:
thanks for  your support .my graduation project about emergency interrupt of the Traffic Signal Lights (i.e ambulance sends light message to the traffic signal lights to set the red sign and after pass the ambulance make it green again )
so i modified that code as when receiver side have that message emrg (emergency) it set red sign and the ambulance send message go to set it green again

imayyami, 2017-06-18:
hi, may i see your hardware sirkuit?