jpiat/arduino #3: Need a little more detailed guide on how to run this code

pratham96, 2016-02-19:
I tried running the code using recommended hardware and connections.
On the transmitting side I used an Arduino mega 2560 and on the recieving side I used an arduino UNO.
I uploaded the code and ran it. I saw the output on the serial monitor of the receiver side.
The only thing I got was:

"Start of receiver program"

Could you please post a little more comprehensive guide to run this project?
Any help is appreciated.

Thank You!

jpiat, 2016-02-19:
What modification did you make to the code to run on the mega ? Does the
emitter led lit up ? Did you read the wiki ?

Karan345, 2016-03-07:
Same problem here. Tried running this on an arduino uno. 

Also if possible can the program be explained a little simpler.

jpiat, 2016-03-07:
Can you describe problem, and the hardware you used (LED color and lens
type clear/diffused). This projest was first build as a quick demo, but
with more and more people trying to use it, i need to come with a better
documentation.

Karan345, 2016-03-08:
Thank you for replying.
I will get back to you soon regarding the further details of the problem in a day or two.

Karan345, 2016-03-09:
1. I am using two arduino unos as  transmitter and receiver. White LEDs on the transmitter side they are regular LEDs and a photodiode on the receiver side.

2. The data is being transmitted through the serial pins of the arduino(Tx and Rx), effectively it is just serial communication without using physical wires. I have not encoded the waveform in any way. Baud rate is 115200.

3. The program is not compiling. It would be helpful if you told us about the files that need to be downloaded to run the program.

4. Also I woud like to implement this project on arduino nanos and it would be helpful  if you could provide us the detailed circuit diagram for the same.

jpiat, 2016-03-09:
This project was never tested (at least by me) with a white LED and a
phototiode so i'am not suprised it does not work. Could you try to run it
with two red, crystal lens LEDs ?  I don't understand item 2 ... Do you sue
my code, or just tied Tx to a LED and RX to a photo-diode ?  Do you use the
Arduino CC IDE (https://www.arduino.cc/) ? The IDE from
http://www.arduino.org/ does not work with this project.

jpiat, 2016-03-09:
I have fixed the project so it can now compile on both IDE (.org and .cc)

2016-03-09 22:40 GMT+01:00 Jonathan Piat <piat.jonathan@gmail.com>:

SandunMadola, 2016-06-13:
Hi, I tried this code in arduino-uno. Emitter Blue Led and receiver Photo Transistor. It's work fine. Got "Hello world <num>". Receiver Blue LED also used. Sometime works fine. Some time gives garbage value. No editing done to the code. I used default code. When you try with Photo Transistor sometime it's not the same way you set LEDs.