jpiat/arduino #5: Special characters

DimMrgrtpls, 2016-03-01:
Hello Jpiat ! 

Firstly, thank you a lot for this tutorial and the code. 

I've built your system with two Arduino UNO. I've managed to transmit something. However, on the receiver serial monitor, there are only special characters (as shown on the attached file)

Have you ever experienced this problem ?

Thank you a lot 

[image]

jpiat, 2016-03-01:
What data re you sending ? Are you running the test code without
modification ?

DimMrgrtpls, 2016-03-01:
I was trying fisrtly to send the "Hello world" by running the code without any modification.

I run one Arduino UNO on one computer, and another Arduino UNO on another computer.

jpiat, 2016-03-01:
This is a bit weird since the receiver should print data to the terminal
only if a correct frame is received. What kind of led do you use (color and
pin its wired to) ?

DimMrgrtpls, 2016-03-01:
I use red brillant LEDs for both emitter and receiver.
On emitter, the LED is connected to pin13, and on receiver on Analog A3.

jpiat, 2016-03-01:
On the receiver side, can you uncomment the line :

//#define DEBUG

to get debug output in the terminal ?

jpiat, 2016-03-01:
> In fact previously I was wrong : on the emitter side, the LED is connected
> to pin2 (Digital).
>
> I've run the debugger on the receiver, and the results is attached to the
> message (and I guess it doesn't help)
>
> [image: capture d ecran 2016-03-01 a 14 17 10]
> <https://cloud.githubusercontent.com/assets/17572089/13435324/827107a8-dfb8-11e5-829a-be5aa510a88a.png>

What is the distance beetween the two leds ? If too close the emitter can
saturate the receiver.

DimMrgrtpls, 2016-03-02:
We tried several distances : from 0cm to 25cm. And it gives each time the same special characters (but it seems to receive something even 25cm away)

DimMrgrtpls, 2016-03-02:
Interesting phenomenon : we've run only the receiver (without any emitter) and in fact it prints the same special characters. So it's probably parasite signals. When you did your experiments, were you in the complete dark ?

We've also tried in a darker place and we managed to have one or two transmitted messages !

jpiat, 2016-03-02:
Do you have the 1Mohm resistor in parallel with the receiver led ?

DimMrgrtpls, 2016-03-02:
Yes we have it !

We did the experiment in a dark place, with a lower resistance for the emitter. And by changing the position of the emitter spot toward the receiver, we managed to find a position where the transmission goes well (there is still a little noise, but like 1 line over 10).

We have also tried with 3 receiver LEDs in parallel. For the moment, we can't really figure out if it works better or not. 

The results are encouraging !

jpiat, 2016-03-02:
I have been running this in plain daylight or in my office lit with halogen
and got no problem. I suspect that the LED characteristic can affect the
receiver. One other thing you can try to play with is the edge threshold in
the receiver.

jpiat, 2016-03-02:
Try to modify EDGE_THRESHOLD to 8 or 15. This will affect distance but can
help reject the noise.

albilamb, 2016-03-04:
Hello jpiat,

I am working on the same project as DimMrgtpls.

We will probably change the LEDs we use when we find better ones. What are the characteristics of the ones you used ?

Concerning the noise, changing the EDGE_THRESHOLD value to 8 did not change so much from the original value of 4. The receiver displayed many messages even when the emitter was not plugged in.

When changing it to 15, the receiver seems not to be so affected by noise : if we turn the emitter off, it receive less information (less messages display). Until what value of EDGE_THRESHOLD should we go ?

Would you recommend using optical filters ?

Thank you a lot!

jpiat, 2016-03-04:
You should first measure the max voltage generated by the receiving LED. If
the voltage is greater than 1.1v you should configure the code to use the
5v reference for the ADC and not the 1.1v (comment #define INT_REF), and
then play with the threshold again.

It would be great to implement an adaptative threshold, but it requires
more work. Optical fiber would be great but then why using visible light ?

jpiat, 2016-03-04:
You need to keep the threshold lower than 128.

2016-03-04 15:03 GMT+01:00 Jonathan Piat <piat.jonathan@gmail.com>:

albilamb, 2016-03-04:
The voltage received by the LED did not exceed 0.3 V so we did not have to use the 5V reference. What voltage did yours receive ? Just to understand if our LED is responsible for the problems we face.

Changing the Threshold (to 100) value made it possible to have a stable communication between the two devices for a distance communication of about 5 cm.

I talked about optical filters in order for our receiver only to "read" the correct wavelength.

We will keep you informed of our advancement.

Thank you for your help.

jpiat, 2016-03-04:
0.3v seems very low ...  can you capture the signal at the receiving led
side with a scope ?

albilamb, 2016-03-30:
[image]

Here is the signal received by the LED. Just a question: why do we have negative values for the tension across the receiving LED ? What voltage did you have ?

Thank you!

jpiat, 2016-03-30:
Your scope must be configured in AC. Configure it in DC and run the capture
again.

DimMrgrtpls, 2016-04-01:
Our scope is configured in DC. 

We are amplifying the signal by 2 with a basic amplifier non inverter and have 0,7 to 1,1V at the output

[image]

However, measuring the voltage over the receiving LED, it doesn't reach 1mV.
Even with a high threshold (127) and in a quite dark environment, there is still plenty of noise.

So we tried configuring the code to use the 5v reference for the ADC. And it works way much better. We are able to transmit the message !

jpiat, 2016-04-01:
Nice ! This is something that really depends on the LED type (color). I'am
not sure of how we could simplify the steps to get the user running with no
need of a scope.

albilamb, 2016-04-11:
Hello jpiat,

Now we would like to work with an RGB LED in order to improve the throughput of our sertup. What did you use as a receiver to be able to "receive" from the different wavelengths ? We tried some phototransistors but unfortunately they were sensitive only to some colors.

Thank you!

jpiat, 2016-04-11:
I used a red LED to receive red, a blue LED to receive blue and a green LED
to receive green ....

albilamb, 2016-04-13:
I was not very clear in my previous post. In your RGB setup, did you use the 5050 RGB LED for reception or three different LEDs ? The RGB LED works well on the emitting side but we are having some trouble making it work on the receiving side..

Moreover, do you have any photodetector reference that you would you recommend for an RGB reception ?

Thanks!

DimMrgrtpls, 2016-05-12:
Hey ! 

I'm giving you some news. 
We now work with an RGB LED and a phototransistor (used as a switch and plugged between +5V and ANALOG INPUT PIN. So not the same setup as with a red LED).
We have tested our phototransistor with an oscilloscope to know its sensitivity. And it's working perfectly !
I have seen that some people try to work with other receiver components than simple LEDs.

We have easily implemented a sending start in the code. We have also tried to transmit data entered in the serial monitor (by uncommenting #TRANSMIT_SERIAL on the emitter side). However nothing happens for the moment. Is the code sending continuously the entered string as it is sending "Hello World" continuously ?

Also, on the receiver monitor, when the sent message is "Hello World", there is anumber appearing next to the message that allow us to know if the transmitting is good or not. However it does not appear with other messages. Where is this coded ?

Thank you !!

imayyami, 2017-06-16:
What can i see hardware you make ? do you wear arduino uno ?

jpiat, 2017-06-16:
For the sending using serial input, the buffer is transmitted only when
there is enough data or when the newline character is received on the
serial input. To achieve this with the arduino IDE serial console, you need
to select the 'Newline' option is the list that is at the left of the
baudrate setting of the terminal.

imayyami, 2017-06-19:
How to circuit in arduino uno?