jpiat/arduino #8: Error when compiling on Arduino Uno using IDE 1.6.8

Cryozenics, 2016-04-16:
I've been trying to compile your codes on my Uno but I've been having problem with LiFi Receiver part. Every time I tried to compile,warnings and errors pop up.Actually a warning pop up for the emitter too but when I tried changing the char* msg  to constant char* msg , it stops popping up(does this change anything though?).However,I don't know what to do on the receiver.Below is the message that pop up.I'm using a Red Led on both Arduino though as I don't have a 120ohm resistor,I'm using 100ohm instead.

C:\Users\User\Downloads\Arduino-master\Arduino-master\LiFiReceiver\LiFiReceiver.ino: In function 'int is_a_word(long int*, int, unsigned int*)':

C:\Users\User\Downloads\Arduino-master\Arduino-master\LiFiReceiver\LiFiReceiver.ino:94:43: warning: left shift count >= width of type [enabled by default]

 #define START_STOP_MASK  ((STOP_SYMBOL << 20) | (START_SYMBOL << 18) | STOP_SYMBOL) //STOP/START/16bits/STOP

                                           ^

C:\Users\User\Downloads\Arduino-master\Arduino-master\LiFiReceiver\LiFiReceiver.ino:98:38: note: in expansion of macro 'START_STOP_MASK'

             if(((*manchester_word) & START_STOP_MASK) == (START_STOP_MASK)){ // testing first position 

                                      ^

C:\Users\User\Downloads\Arduino-master\Arduino-master\LiFiReceiver\LiFiReceiver.ino:94:66: warning: left shift count >= width of type [enabled by default]

 #define START_STOP_MASK  ((STOP_SYMBOL << 20) | (START_SYMBOL << 18) | STOP_SYMBOL) //STOP/START/16bits/STOP

                                                                  ^

C:\Users\User\Downloads\Arduino-master\Arduino-master\LiFiReceiver\LiFiReceiver.ino:98:38: note: in expansion of macro 'START_STOP_MASK'

             if(((*manchester_word) & START_STOP_MASK) == (START_STOP_MASK)){ // testing first position 

                                      ^

C:\Users\User\Downloads\Arduino-master\Arduino-master\LiFiReceiver\LiFiReceiver.ino:94:43: warning: left shift count >= width of type [enabled by default]

 #define START_STOP_MASK  ((STOP_SYMBOL << 20) | (START_SYMBOL << 18) | STOP_SYMBOL) //STOP/START/16bits/STOP

                                           ^

C:\Users\User\Downloads\Arduino-master\Arduino-master\LiFiReceiver\LiFiReceiver.ino:98:59: note: in expansion of macro 'START_STOP_MASK'

             if(((*manchester_word) & START_STOP_MASK) == (START_STOP_MASK)){ // testing first position 

                                                           ^

C:\Users\User\Downloads\Arduino-master\Arduino-master\LiFiReceiver\LiFiReceiver.ino:94:66: warning: left shift count >= width of type [enabled by default]

 #define START_STOP_MASK  ((STOP_SYMBOL << 20) | (START_SYMBOL << 18) | STOP_SYMBOL) //STOP/START/16bits/STOP

                                                                  ^

C:\Users\User\Downloads\Arduino-master\Arduino-master\LiFiReceiver\LiFiReceiver.ino:98:59: note: in expansion of macro 'START_STOP_MASK'

             if(((*manchester_word) & START_STOP_MASK) == (START_STOP_MASK)){ // testing first position 

                                                           ^

Cryozenics, 2016-04-20:
I got it working now after attaching an additional 220ohm resistor.Though the output have a number counter at the back example "Hello World0" and it increases till 9 before going back to 0.It's good enough for me though as a test.

ASHNR, 2016-04-21:
Hi,
I'm working on the project as well
what was the distance between the emitter and receiver?
did you use A0 or A3 as the analog input?
I just get " Start of the receiver program"
and nothin else.
any suggestions based on your experiences?

Cryozenics, 2016-04-22:
Sorry for the late reply.I used A0 as recommended.I forgot to add that I already swapped the red LEDs with clear blue LEDs instead as recommended too. The distance I tested with blue was around 5cm(clear red should give better distance).Before I just used 1 Mohm resistor for the receiver,100ohm resistor on the transmitter and only received garbage bits but after I attached another 220ohm resistor on the receiver,I started to receive the msg "Hello World" followed by 0 till 9 in loop.

ASHNR, 2016-04-22:
Thanks for the reply!
Sure it's A0 and not A3?
cuz both are mentioned in the explanation
and did you use the 220 ohm in series to 1Mohm or parallel.
which pin did you use at the transmitter 13 or 2?

Cryozenics, 2016-04-22:
Yup it's attached to A0.I put the 220 ohm parallel to the 1Mohm.For transmitter I used pin 2.

Yuejing1992, 2017-03-20:
Hi,
I encountered same errors as yours. Did you do any modification on the code besides parallel a 220 ohm resistor with 1Mohm resistor? If not, could you please explain why this happened?

Yuejing1992, 2017-03-21:
Hi, for those who encountered as errors. You can change start symbol and stop symbol data type to uint32_t since by default is 16 bits only.

imayyami, 2017-06-18:
hi, may i see your hardware sirkuit?

sunilkumarampolu1998, 2018-01-14:
Hello imayyami,
did you find circuit diagram?if yes,please do share it