jpiat/arduino #9: pin connection in transmitter

ASHNR, 2016-04-18:
Hi jpiat,
had a quick question
The instructions in a demo say that the led must be connected to pin 2
but i think you've used pin 13 in the code
which one of them is right?

jpiat, 2016-04-18:
The code uses pin 2, but you can run on pin 13 by uncommenting the appropriate section saying : " These defines are for a LED connected on D13" and commenting the section :

#define OUT_LED() DDRD |= ((1 << 2))
#define SET_LED() PORTD |= ((1 << 2))
#define CLR_LED() PORTD &= ~((1 << 2))

ASHNR, 2016-04-19:
Thank you so much!
I've connected the circuit using blue clear led and the transmitter glows as soon as the code is loaded.
However, the receiver does not show the serial data. It just says "Start of receiver program"  The led's are placed like 2 cm apart.
I've programmed the arduino pro mino for 5V is that ok?
or should it be programmed for 3.3V..

jpiat, 2016-04-19:
The voltage of your arduino pro mini depends on the version you have, not
my program. Can you try to put the two arduino further appart (15cm) and
try to align the LED ?

ASHNR, 2016-04-19:
No, I'm actually talking about programming the pro-mini with an FTDI. 
Right now i'm receiving junk values at the receiver.

ASHNR, 2016-04-21:
can I use a photo transistor instead of an led?

ASHNR, 2016-04-21:
Sorry to be bothering you!
but does the emitter work on negative logic?
because it switches on even before any message is transmitted in the serial port
(after uncommenting #define transmit serial)

jpiat, 2016-04-21:
Even if there is nothing to transmit, the emitter will be on. This project
mimic a ceiling light being use for broadcasting data, you would not want
your lights to turn off if there is nothing to transmit. The light appears
as always on for the user while it may or may not transmit data.

ASHNR, 2016-04-21:
Thanks for that!
should A0 or A3 be used at the receiver?
u've mentioned both in the writeup....

jpiat, 2016-04-21:
Its ADC3 :

#define SENSOR_PIN 3

ASHNR, 2016-04-23:
Hi jpiat,
I've tried out all the possible kind of led's and distances mentioned but i don't receive the any output :(
At times I get some junk values at the receiver but at a very slow rate.
Dissapointed :(
any suggestions?
when I uncommented the #DEFINE DEBUG i get junk numbers and values at the receiver.
Any help would be appreciated.

jpiat, 2016-04-23:
I am on vacation right now and I don't have access to the code. Can you try
to look for a define in the receiver code to use internal reference for the
ADC and comment it.

ASHNR, 2016-04-23:
Oh...ok!
Sorry to disturb!
I've tried that too. Doesn't work...

imayyami, 2017-06-18:
Could i see hardware sirkuit being worked for lifi because i am working on the same project but troubled in hardware sirkuit?