jpiat/arduino #15: Increase the number of characters

amirezz, 2016-09-01:
Dear jpiat "
Hope you are doing well.
thanks for your great code it is very helpful.
i was wondering of there is any way to increase the number of characters send every time.
thanks in advance

carlosascari, 2016-09-02:
Hello @amirezz 
*I'm following this project, so I got notified of this issue and felt like passing my 2 cents*

I believe the fastest way to increase the speed is by turning the delay from `10ms` to `1ms`
[Here is the line](https://github.com/amirezz/arduino/blob/master/LiFiEmitter/LiFiEmitter.ino#L162) for the transmitter.

Here, the [delay()](https://www.arduino.cc/en/Reference/Delay) function is explained.

If however you are looking for a faster rate, there is a [delayMicroseconds()](http://www.arduino.org/learning/reference/delaymicroseconds) function.
Although, there seems to be [some caveats](http://forum.arduino.cc/index.php?topic=46452.0)

I suppose anything faster that remains accurate cannot be done with arduino alone, you will need a *Driver*, a separate module whose purpose is to transmit/receive pulses of light a very high rates, and interfaces with the arduino.

sayedhatem, 2016-09-04:
Dear @carlosascari 
Thanks for your interest . but i don't care about the speed now 
.I use the serial monitor to transmit the data so i need to send more than 32 char each time
and here in the comments line that i can't 
0x02 : STX start of frame
N times Effective data excluding command symbols, with N < 32
0x03 : ETX end of frame
any ideas. thanks in advance

amirezz, 2016-09-04:
Hellow @carlosascari 
Thanks for your replay . that was very helpful