jpiat/arduino #16: not receiving

fadvictim, 2016-09-07:
Hi!

First of all congrats on this project and thank you for sharing!
I am using the original code with no modifications  and .cc IDE (although the new .h file solve that, just in case)

As in hardware:

2 arduino nano 328
2 5mm clear red LED
1 Mohm (Rx)
120 ohm (Tx)

emitter is in D13 and always on as espected but receiver in A0 is getting nothing no matter what distance (0-50cm) or light conditions I try I only get "Start of receiver program"

I have turn the debug if that is any help

Thanks!

[image]

fadvictim, 2016-09-07:
Was just thinking.. do the LEDs need to be in line of sight (top to top)?

luizmendes, 2016-10-25:
Hi @fadvictim ,

About the LED, in my case, the emitter needs to be directly pointed to the receiver.  

About the pins, if your receiver is on pin A0, the receiver's code should be changed, because in the way it is, the receiver is connected to analog pin 3.

Here is the piece of code that should be changed:

`//This defines receiver properties
#define SENSOR_PIN 3
#define SYMBOL_PERIOD 500
#define SAMPLE_PER_SYMBOL 4`

Just change it to "#define SENSOR_PIN A0", and see if it works. This worked for me.
Hope it helps you.

Yuejing1992, 2017-03-21:
Hi, i am using A3 for my receiver, but the received signal are too weak. The H-L difference is only about 50mv. Did you encounter same problem? I used photodiode instead of LED coz it has better performance.

deepak0003, 2017-04-08:
after changing that pin configuration I am not receiving the transmitted data.please help me out as it mine final year project.

luizmendes, 2017-04-09:
Hi @deepak0003 

Before that, were you receiving any data?

You can get the original code and put the sensor on pin A3. It should work.

deepak0003, 2017-04-09:
thanks @luizmendes  so promptly replying

YES. I am receiving some garbage data.I have checked the serial moniter for transmitter .it was sending the same data frame as in the code.But the reciever isn't getting that data.insted some garbage value is coming.

have you got the proper output result.

deepak0003, 2017-04-09:
HI @jpiat 
Sir,I am not able to receive the transmitted message of hello world in LIFI project.I am receiving some garbage value.
Should I use photodiode instead of led  parallel with resistor.

jpiat, 2017-04-09:
Do you have a picture of your setup ? What kind of led do you use
(clear/diffused, color )?
Jonathan Piat

<https://github.com/jpiat/arduino/issues/16#issuecomment-292772465>, or mute
the thread

deepak0003, 2017-04-09:
hi
components are
Transmitter side
1.white Led
2. Arduino UNO r3
3.resistor with 270 ohms in series with led 
4.Led is connected to digital pin 13(uncommented the line in the code for making digital pin 13 for output)
5.postive pin of led is connected to pin 13 and the negative end to ground.

Receiver side
1.Arduino UNO r3
2.White Led with 1M ohms in parallel
3.Led reverse biased connected to analog input pin 3 and ground

while checking the output in serial monitor baud rate set to 115200  and I checked with 1200 baud rate too and much more.
still, i am not getting the required output.please help me out and if possible give your skype id or facebook id through which I can do video conferencing.
I know it is too much to ask but I am very much in need of any type help.

deepak0003, 2017-04-09:
hi @jpiat 
components are
Transmitter side
1.white Led
2. Arduino UNO r3
3.resistor with 270 ohms in series with led
4.Led is connected to digital pin 13(uncommented the line in the code for making digital pin 13 for output)
5.postive pin of led is connected to pin 13 and the negative end to ground.

Receiver side
1.Arduino UNO r3
2.White Led with 1M ohms in parallel
3.Led reverse biased connected to analog input pin 3 and ground

while checking the output in serial monitor baud rate set to 115200 and I checked with 1200 baud rate too and much more.
still, i am not getting the required output.please help me out and if possible give your skype id or facebook id through which I can do video conferencing.
I know it is too much to ask but I am very much in need of any type help.

jpiat, 2017-04-09:
Hi, this won't work with white led (since a white led is a blue one coated
with phosphor). If you want yo transmit with white light, use a RGB led and
modulate only one component.

Jonathan Piat

deepak0003, 2017-04-09:
ok thanks for that i will let you know the output once i use RGB led.
thanks for the insight

deepak0003, 2017-04-09:
hi @jpiat 

Which type would be better Diffused or Clear red led .
please could you tell me which led you used and resistor so that i would be sure that output will come.
 
for reciever end ,arduino internal voltage should be 5 or 3 or 1.

jpiat, 2017-04-09:
As stated on the wiki, use only clear led (works better with high
luminosity one). For a red led, a 220ohm resistor will be fine.
Jonathan Piat

sunilkumarampolu1998, 2018-01-14:
hello jipiat,

will your code work on Arduino UNO. if yes, any modifications required?