jpiat/arduino #17: Which LED ?

xelaco, 2016-11-04:
This code is very interesting for me.
I tested with the red and the blue LEDs of the official Genuino starter kit.
I think my LEDs have not enough power because I must put the LED at about 1 mm. 
Could you give me a reference of usable LEDs for this project,  please?
Thanks in advance. 
Sorry for my very bad English...  I prefer writing in French...

jpiat, 2016-11-04:
Bonjour,

le code fournit n'est pas super stable donc vous pourriez avoir besoin de mon aide pour le faire fonctionner. N'importe quel LED rouge en boitier crystal marche bien (led 5mm de préférence) il faut juste s'assurer de l'alignement de l'emetteur et du récepteur.

xelaco, 2016-11-04:
Bonsoir,
Merci pour la réponse rapide ! Mes LEDs ont un boitier rouge, donc je pense qu'elles ne conviennent pas...
Je vais en acheter d'autres.
J'ai quelques questions à vous poser mais il faudrait que je vous donne des informations que je ne souhaite pas rendre publiques. Ainsi, je préférerais que nous communiquions par email si cela ne vous dérange pas, mon adresse étant visible sur mon profil.
En vous remerciant par avance pour votre réponse,
xelaco

knwachuk, 2017-03-01:
Good evening,

I did a few test using available LEDs and found that the diffused LED would work. Here are my preliminary test cases:

 Emitter LED | Receiver LED | Functional Distance 
--------------|--------------|--------------------
Diffused blue (3mm) | Clear white (5mm) | 10cm
Clear white (5mm) | Diffused blue (3mm) | 10cm
Diffused blue (3mm) | Diffused blue 30cm | Not sure but functional
Clear white (5mm) | Clear white (5mm) | Not sure but functional

I do not know if the functional LED list has been updated, but I think it may help others. Thank you very much for your wonderful work.

knwachuk

jpiat, 2017-03-01:
Hi,

Thanks a lot, will add to the wiki. Do you think their would be any
interest in a custom shield for this demo (with a known to work well
emitter and receiver) ?

Jonathan Piat

knwachuk, 2017-03-01:
Yes.

If you are asking that I should create one, definitely yes.

If you are asking if people would be interested, definitely yes. 

knwachuk

jpiat, 2017-03-01:
We could collectively create one to ease the development and maybe bring
what is just a demo into a library to be integrated in applications.
Jonathan Piat

knwachuk, 2017-03-01:
Then I am definitely interested!

knwachuk