jpiat/arduino #18: util/atomic.h library not available

shivanksharma91, 2016-11-14:
hello

I was trying to go through this code, then i realized that this "util/atomic.h library is not available in arduino.
Can anyone help me in this? Where to find i and what does this library do?

jpiat, 2016-11-14:
What version of Arduino IDE are you using ? What Arduino board are you
targeting ?

shivanksharma91, 2016-11-15:
I am using arduino uno and Arduino IDE ver 1.6.13

jpiat, 2016-11-15:
I just tried and it worked with arduino 1.6.12 which is the latest stable
version. Can you make sure that you have configured the IDE for arduino UNO
(Arduino/Genuino Uno should appear at the bottom left of the IDE window).

Regards,

Jonathan Piat

shivanksharma91, 2016-11-21:
I don know, i followed each instruction correctly,but when i start the serial plotter i just see the "Start of receiver program" message and nothing else happens.
If possible can you please explain me how the Manchester encoding & decoding is being implemented in the provided code?

Regards

Shivank Sharma

closed 2020-11-21 by shivanksharma91