jpiat/arduino #20: Combine modulation with Manchester?

Jozhxy, 2017-02-14:
Hi, I've tried out your code, works fine with x9 LEDs to photodiode at other side.
Since Arduino UNO can't run the code in parallel. Is it possible to combine ASK into it without modifying the circuit at few hundreds Hz or kHz and then demodulate again?

jpiat, 2017-02-14:
In its current state I don't see how to do this. If your goal is to
increase throughput, you could use rgb LED and have up to three time the
bandwidth. You can also probably use a higher transmitter rate if you are
using a photodiode
Jonathan Piat

Jozhxy, 2017-02-14:
Thank you for your quick respond. I found the hot spot at SYMBOL_PERIOD 2400 & changed the threshold which is still perfect square wave on the oscilloscope, unfortunately my photodiode won't detect any faster than that. But I'm glad the LED is above the threshold of human eye to perceive.
Thank you once again 👍

jpiat, 2017-02-14:
Can you give the reference for the photodiode you are using ?
Jonathan Piat

Jozhxy, 2017-02-14:
Oh sorry, it's a phototransistor. I use a macro lens to magnify in order to improve the signal.
http://www.vishay.com/docs/81321/tept5700.pdf
Bought it from: http://my.element14.com/vishay/tept5700/photo-transistor-npn-570nm-5mm/dp/1497676

jpiat, 2017-02-14:
> transistor-npn-570nm-5mm/dp/1497676

Nice ! I'll try to replicate your setup. Using a LED as receiver does not
work reliably.
Jonathan Piat

Jozhxy, 2017-02-14:
Cool.
Yea, because I need to match the wavelength with the transmitter, most receivers are at IR range.
To achieve a standard communication system, other than modulation, what else could be implement into this project? (If arduino could handle it)

jpiat, 2017-02-15:
I have a TEMT6000 on hand, i'll see if it could work. Interesting things to
implement would be :
- CRC computation (to discard incomplete frames)
- auto-sync (detect the symbol frequency)
- design an arduino shield to ease experiments (some people had trouble
with the wiring and the use of a wrong led as receiver)
- spectrum spreading (to have multiple channels)

....

shivanksharma91, 2017-03-11:
@Jozhxy Can you help me with this? please shivanksharma91@gmail.com

Jozhxy, 2017-03-11:
@shivanksharma91 For this case, only IR is fast enough to modulate cuz it's the standard.
LED will dim as it doesn't have much time to stay fully on or off. There's only few that manufactures high switching LED & phototransistor/diode which are very costly.
You can use function generator as an external modulation into [AND transistor](http://hyperphysics.phy-astr.gsu.edu/hbase/Electronic/trangate.html) or if you can code a fixed modulation frequency while sending data in Arduino, maybe extract IR library and modify it.

shivanksharma91, 2017-03-11:
@Jozhxy thanks for the quick respnse, i need help in using photodiode at receiving end. If i just replace the led with photodiode on the receiving end and upload the code, will it work? Since i cannot really understand the code, i am new to it.

Jozhxy, 2017-03-11:
The code sends out by frame, loop Hello Word 1 to 10.
Try it short distance, adjust to lower speed first in the code. Get a variable resistor to adjust photodiode sensitivity like IR receiver circuit.
See other issue posted before for further info of the code