jpiat/arduino #21: Extract value of 1 and 0 at receiver

bl1229, 2017-02-16:
Hi, I used your code to get communications between an LED and a photodiode. I am now trying to implement a positioning algorithm and I need to be able to find the sensor reading that corresponds to receiving a 1 and a 0  for distance estimation purposes. Do you know if it will be possible to do this with this code?
Thank you

jpiat, 2017-02-16:
I am not sure to understand ... Would you have many transmitters and the
refeiver would triangulate its position based on the transmitter it sees ?
Jonathan Piat

bl1229, 2017-02-16:
Yes I have 4 transmitters to one receiver. But it sees multiple transmitters at the same time and only one transmitter sends data at any one time (when not sending data the rest transmit at a constant high value - no Manchester). Based on the difference in light intensity between receiving a 1 and a 0 the Rx will find its distance from each of the Tx. These distances then used for triangulation

jpiat, 2017-02-16:
You cannot use the light level for triangulation since it will depend on
many factors (orientation or ambient light level) that have nothing to do
with localization. I have no idea on how you can triangulate the receiver
position using VLC. You can try GPS like localization but it would require
your transmitter to be accurately synchronized, and this won't work at room
scale. What you can do is use multiple receiver and try to triangulate only
based on the emitters you see.
Jonathan Piat

bl1229, 2017-02-17:
The receiver is always on a plane parallel to the transmitters so orientation angle is not a problem. The distance from the Tx to Rx is related to the difference in light level from a 1 and a 0 so the ambient light will not be a problem because it is constant, I am conducting experiments in the dark anyway so the only ambient light in the system is from other transmitters. In your code do you calculate the light level associated with a 1 and 0 or is just based on the assumption that if the light level is higher than the last bit by the threshold it must be a 1 and lower than the last bit it must be 0?
Thank you