jpiat/arduino #22: variety words transmission

JeasonZzz, 2017-02-24:
Dear Jpiat
Your code could run properly but we tried to transfer variety of words instead of hello world. We erase // before #define TRANSMIT_SERIAL and because of  the sentence if(com_buffer_nb_bytes >= 32 || c == '\n'), if we press Enter on the keyboard the inputted sentence should be transferred. But the situation now is we could only transfer data if we input more than 32 words.

jpiat, 2017-02-24:
If you had the new line character '\n' the message will be transmitted even
if its shorter than 32.
Jonathan Piat

JeasonZzz, 2017-02-24:
we tried the new line character '\n', but it could not help. The words can only be transmitted if the number of words accumulated to 32. And the part more than 32 words can not show until the next 32 words are inputted.

jpiat, 2017-02-24:
Hiw do you send '\n' to the arduino ?
Jonathan Piat

JeasonZzz, 2017-02-24:
Now it works! Thanks!