jpiat/arduino #23: Emitter LED

4cho, 2017-02-24:
Dear Jpiat,

How did you connect wires to the emitter LED and how many arduino mini is need in emitter circuit? I am using the same emitter LED like you but i can't figure out how to connect to the arduino. Actually, I cant see any pin on emitter LED.

jpiat, 2017-02-24:
Each code (receiver code and emitter code) has a ASCII drawing on how to
connect the led at its top. For the emitter, you'll need one arduino, wired
to a led through a current limiting resistor, and for the receiver, another
arduino with a led wired to an analog port with a parallel led.
Jonathan Piat

4cho, 2017-02-24:
First of all, thanks for your replay, Jonathan. I saw the ASCII drawing and it is really helpful. For the emitter LED, did u drilled or open the LED to connect the wire inside because I bought the LED which is AC power supply since I can't find another emitter LED like the one you are using.

jpiat, 2017-02-24:
You first need to use a plain old 5 or 3mm led dot not start with a LED
bulb. To use a LED bulb, i had to make major modifications to it.

4cho, 2017-02-24:
OH... I see. I was trying to figure out how to use that led bulb. :laughing: Thank you for your advise, sir. :+1:

inwan96, 2018-04-27:
@4cho have you found a way to use a led bulb in transmitter ?
because my project use a led bulb sir @jpiat