jpiat/arduino #24: Mega 2560

Yuejing1992, 2017-03-03:
Hi,
I would like to ask if i use Mega 2560 instead of mini pro any modification i need to make? Besides hardware connection issue and change the timer1 function to timer5. I assume since mega 2560 have a more powerful microprocessor than pro mini, will it make sense that it can somehow achieve transmission speed than pro mini?

jpiat, 2017-03-03:
Hi,

i guess that the modification will only concern Timer and ADC (i don't use
the arduino library for the ADC but straight register manipulation). The
transmission does not depend on the CPU speed, its mainly a matter of using
a LED that is not designed to act as a receiver. Using a proper photodiode,
higher rates could be achieved.

regards,

Yuejing1992, 2017-04-11:
Hi,

I have successfully integrated your code with a white LED  with an LED driver circuit and also added in a amplifier circuit for the photo-diode in the receiver side also. Anything works fine.
I am trying to increase the bit-rate by decrease the symbol period to 250 microsecond, but the receiver side only receive "Start" (Not an complete sentence of "start of the program").
Would like to ask why...

TheLastAvenger, 2017-04-14:
@Yuejing1992 Hey, i am trying it on the Arduino Uno and beside the resistors and some changes in the code, I get the LED to blink, can you share your setup on MEGA?, I also want to try.

jpiat, 2017-04-18:
I never tested this code for bitrates higher than 1kbit. Doubling the bit-rate (with 250µs symbol period) may be too fast. Did you have a look at the signal on the receiving side ?