jpiat/arduino #26: Emitter and Receiver in one file as Transceiver?

qulmanaqim, 2017-03-16:
I have tried making a transceiver using both the emitter and receiver codes in one file but it didn't work as expected. I have tried using two interrupts using timer1 and timer3 for sample_edge and bit_emit_half but the received data was kind of scrambled. Do you have any advice in making a transceiver using the codes?

deepak0003, 2017-04-09:
hi @qulmanaqim 
did you find yet that transceiver?

qulmanaqim, 2017-04-30:
No, I haven't. Do you have any suggestion, @deepak0003?

qulmanaqim, 2017-05-02:
@jpiat , what should I change for this to work?
```
#include <TimerOne.h>
#include <TimerThree.h>
#include "receiver_types.h"
#include <util/atomic.h> //atomicity to prevent interrupt

#define DEBUG_HEX
//#define DEBUG_DATA
//#define DEBUG_ANALOG
//#define INT_REF /* Comment this to use AVCC reference voltage. To be used when the receiver LED generate low levels */
//#define TRANSMIT_SERIAL //uncomment to use serial mode

//This defines receiver properties
const int SENSOR_PIN = 0; //the analog pin number of the LED sensor

const int ledPin = 53; //the digital pin number of the LED emitter
const int powerPin = 11; //button power?
const int togglePin = 12; //button toggler?

#define SYMBOL_PERIOD 1000
#define SAMPLE_PER_SYMBOL 4
#define WORD_LENGTH 10 // a byte is encoded as a 10-bit value with start and stop bits
#define SYNC_SYMBOL 0xD5 // this symbol breaks the premanble of the frame
#define ETX 0x03 // End of frame symbol
#define STX 0x02 //Start or frame symbol 

enum receiver_state rframe_state = IDLE ;

// global variables for frame decoding
char eframe_buffer[38] ;
int eframe_index  = -1 ;
int eframe_size = -1 ;
char rframe_buffer[38] ;
int rframe_index  = -1 ;
int rframe_size = -1 ;

//state variables of the manchester encoder
unsigned char bit_counter = 0 ;
unsigned short data_word = 0 ;  //8bit data + start + stop
unsigned char half_bit = 0 ;
unsigned long int manchester_data ;

void to_manchester(unsigned char data, unsigned long int * data_manchester){
  unsigned int i ;
 (*data_manchester) = 0x02 ; // STOP symbol
 (*data_manchester) = (*data_manchester) << 2 ;
  for(i = 0 ; i < 8; i ++){
    if(data & 0x80) (*data_manchester) |=  0x02  ; // data LSB first
    else (*data_manchester) |= 0x01 ;
    (*data_manchester) = (*data_manchester) << 2 ;
    data = data << 1 ; // to next bit
  }
  (*data_manchester) |= 0x01 ; //START symbol
}

//emitter interrupt
void emit_half_bit(){
     if(manchester_data & 0x01){
        digitalWrite(ledPin, HIGH);
     }else{
        digitalWrite(ledPin, LOW);
     }
     bit_counter -- ;
     manchester_data = (manchester_data >> 1);
     if(bit_counter == 0){   
        //is there still bytes to send in the frame ?
        manchester_data = 0xAAAAAAAA ; // keep sending ones if nothing to send
        if(eframe_index >= 0 ){
          if(eframe_index < eframe_size){
[... 279 lines cut ...]
                 received_data &= ~0x01 ;
                 digitalWrite(ledPin, LOW);
             }
    }
    received_data = received_data & 0xFF ;
    #ifdef DEBUG_HEX
      digitalWrite(ledPin, HIGH);   
      Serial.print(received_data & 0xFF, HEX);
      //Serial.print(", ");
    #endif
    #ifdef DEBUG_DATA
      Serial.println((char) received_data);
    #endif
    new_word = 0 ;
    if((byte_added = add_byte_to_frame(rframe_buffer, &rframe_index, &rframe_size, &rframe_state,received_data)) > 0){
      //rframe_buffer[rframe_size-1] = '\0';
      digitalWrite(ledPin, HIGH);   
      Serial.println(&(rframe_buffer[1]));
    }
    //if(rframe_state != IDLE) Serial.println(received_data, HEX);
  }
  }
}

```